arduino/lab-micropython-installer #8: Installer Not Working with Nano RP2040

robertgallup, 2023-07-21:
When I start the installer, it detects the Nano RP2040, but when I click the install button, it downloads the firmware, but then hangs. This is in non-bootloader mode.

If I switch to bootloader mode (RPI-RP2), the installer does not detect the board.

It's likely I'm missing something, but ?

sebromero, 2023-07-25:
Hi @robertgallup 
Thank you for reaching out. As you correctly pointed out, the board is not detected when in bootloader mode and this is because the RP2040 does not expose a serial port when in bootloader mode. The installer currently relies on a the serial port to be available.
It should work however in "normal" mode. Can you run the installer from the command line and post the output here?
e.g. `$"/Applications/MicroPython Installer.app/Contents/MacOS/micropython-installer"`
Thanks!

robertgallup, 2023-07-25:
Hi @sebromero. Thanks for your reply! The output from running the installer from the command line is:

`/Applications/+MAKE/MicroPython\ Installer.app/Contents/MacOS/micropython-installer
feedURL https://update.electronjs.org/arduino/lab-micropython-installer/darwin-x64/1.2.0
requestHeaders { 'User-Agent': 'update-electron-app/2.0.1 (darwin: x64)' }
checking-for-update
update-not-available
🔍 Finding latest firmware for board 'ARDUINO_NANO_RP2040_CONNECT' ...
🔗 Firmware URL: https://downloads.arduino.cc/micropython/ARDUINO_NANO_RP2040_CONNECT-20230426-v1.20.0.uf2
🌐 Downloading firmware ...
👀 Device found: Nano RP2040 Connect at /dev/tty.usbmodem11101
📤 Sending REPL command: import sys; print(sys.implementation.version)`

sebromero, 2023-07-26:
@robertgallup Hmm, you're probably already running an older version of MicroPython on the board that did not have an api for `sys.implementation.version`. Can you tell me which version that is? I'll try to replicate it.
That said, can you (after making a note of the current MicroPython version) try to flash a blink sketch from the Arduino IDE and then run the MicroPython installer again? That should work. You'll only have to do this once. Any future version of MicroPython should not cause this problem again.

robertgallup, 2023-07-26:
@sebromero So, I did have MicroPython on the board already, but it was the current version, 1.20.0 from 2023-04-26.

Here's what I did next:

1. I flashed it with blink from the Arduino IDE 2.0.4 (later versions of the IDE won't load example sketches for me--another issue)
2. The MicroPython Installer failed again and suggested I put it into bootloader mode
3. I put the board into bootloader mode and ran the Installer again from the command line
4. This time there was success!
5. BUT, then I tried MicroPython Installer again and it failed the same as before.

It seems maybe there another issue?

robertgallup, 2023-07-27:
OK. I think I figured out the problem. I had a boot.py that just printed numbers in a loop across the serial connection. When I renamed that to no-boot.py to disable it. The installer seems to work just fine. Yay! :-)

sebromero, 2023-07-27:
@robertgallup Great! I will add a section to the readme file for anyone else running into those issues. Thanks for investigating! 🙏