arduino/lab-micropython-installer #10: Message "Error invoking remote method 'on-file-selected' : Error"

tacker66, 2023-07-24:
Message "Error invoking remote method 'on-file-selected' : Error" 
appears when choosing a firmware file manually or when drag-dropping.

sebromero, 2023-07-25:
Hi @tacker66 
Can you please give me some more information:
- What operating system do you use?
- On which board are you trying to flash MicroPython?
- Which firmware file are you trying to install via drag and drop?
Thanks!

tacker66, 2023-07-25:
Hi @sebromero 
I am on Win10 and try to flash an Arduino Nano ESP32 with 'firmware.bin' built from https://github.com/micropython/micropython/tree/master/ports/esp32/boards/ARDUINO_NANO_ESP32 on a Linux machine.

sebromero, 2023-07-25:
Thanks for your quick reply. So to flash that firmware, the board indeed would need to be in native bootloader mode. We'll add that shortly. For now I recommend to use the "standard" way which is to not put the board in bootloader and just click "Install MicroPython".

venugudavalli, 2023-07-26:
> Thanks for your quick reply. So to flash that firmware, the board indeed would need to be in native bootloader mode. We'll add that shortly. For now I recommend to use the "standard" way which is to not put the board in native bootloader and just click "Install MicroPython". That way it will install the application image (a special build) which should run fine.

Hi sebromero, I am new to Arduino, could you please let me know, how to do it in standard way. Thanks.

sebromero, 2023-07-26:
Hi @venugudavalli The standard way is to just plug in the board and click on "Install MicroPython". As pointed out in #11 , on Windows you need to first install the drivers for the board.

tacker66, 2023-07-31:
Hi @sebromero, this now works correctly with v1.2.1. Thx :-)

closed 2023-08-01 by sebromero