arduino/lab-micropython-installer #9: Error "Installing from native bootloader is not supported yet."

tacker66, 2023-07-24:
Hi, I am getting "Installing from native bootloader is not supported yet." on a Nano ESP32.

Is this planned resp. is there a possibility to restore the original bootloader?

petrosxen12, 2023-07-24:
Following up on this, what is the suggested method on flashing the Nano ESP32 board without the Micropython installer? 

As I understand from the [official Arduino tutorial](https://docs.arduino.cc/micropython-course/course/installation#step-by-step-installation) for installing MicroPython on Nano ESP32 it is assumed that the MicroPyton installer should work without issues. Is this the case or is the above error due to "...not be flashed when in native bootloader mode. " (quoting from "Limitations" of the v.1.2.0 release)

Can the tutorial documentation be updated to reflect the limitations of the installer if the board is not yet supported?

sebromero, 2023-07-25:
Hi @tacker66 and @petrosxen12 !
At the moment flashing the application image is the only supported mode which means that you should not put the board into native bootloader mode to install MicroPython. Just connect the board and click install.
That said, we'll very soon add support also for native bootloader mode.
Documentation is being updated, thanks for pointing this out!

tacker66, 2023-07-25:
Just a hint for others who have the same problem:

Current Thonny 4.1.1 (https://github.com/thonny/thonny/releases) supports flashing firmware files from
https://github.com/micropython/micropython/tree/master/ports/esp32/boards/ARDUINO_NANO_ESP32
to Arduino Nano ESP32 in native mode :-)

sebromero, 2023-07-25:
@tacker66 To explain a few things: the native bootloader never gets overwritten, it's in ROM. The firmware that this tool currently installs is the same as you would see on [https://github.com/micropython/micropython](https://github.com/micropython/micropython/tree/master/ports/esp32/boards/ARDUINO_NANO_ESP32) without the second stage bootloader, and relocated to a different address. That makes it compatible with Arduino sketches without having to restore anything or manually enter the bootloader. That way you can easily switch back and forth between Arduino Sketches and MicroPython if you wish.

sebromero, 2023-07-28:
@tacker66 We just released a new version that adds support for flashing in ROM bootloader. Please give it a try if you can and let me know if it works so we can close this issue. https://github.com/arduino/lab-micropython-installer/releases/tag/v1.2.1

tacker66, 2023-07-31:
Hi @sebromero, this now works correctly with v1.2.1. Thx :-)

closed 2023-08-01 by sebromero