arduino/lab-micropython-installer #11: Flashing firmware Aurdino Nano ESp2

venugudavalli, 2023-07-26:
I am trying to flash the Micro Python firmware on new Aurdino Nano ESP32 using Micro Python installer. Keep getting this error 
Error invoking remote method 'on-install': Error running dfu-util: 'Command failed: "C:\Users\venug\AppData\Local\MicroPythonInstaller\app-1.2.0\resources\app\node_modules\firmware-flash\bin\win32\ia32\dfu-util.exe" -a 0 -d 0x2341:0x0070 -D "C:\Users\venug\AppData\Local\Temp\nanoesp32_jul14_micropython.bin" -Q Cannot open DFU device 2341:0070 found on devnum 20 (LIBUSB_ERROR_NOT_FOUND) No DFU capable USB device available '
I am using C type USB cable to connect. Board is detected by Installer.

venugudavalli, 2023-07-26:
This is on Windows 11.

sebromero, 2023-07-26:
Hi @venugudavalli ! It looks like the driver for this board is not installed. Unfortunately on Windows a driver is needed.
You can install it by installing the core in the Arduino IDE. In the board manager search for "Nano ESP32" and install the core. Detailed instructions on how to do this can be found [here](https://docs.arduino.cc/tutorials/nano-esp32/getting-started-nano-esp32#install-board-package).
When you're done try again using the MicroPython Installer tool. Please let me know if it works.

venugudavalli, 2023-07-26:
> Hi @venugudavalli ! It looks like the driver for this board is not installed. Unfortunately on Windows a driver is needed. You can install it by installing the core in the Arduino IDE. In the board [...]

Hi sebromero. Yes I did it. It works now. Thanks

closed 2023-07-26 by sebromero