arduino/lab-micropython-installer #13: Error invoking remote method 'on-install'

adengle, 2023-07-27:
Brand new Arduino Nano ESP32. Attempting to follow the instructions from the MicroPython 101 course (https://docs.arduino.cc/micropython-course/course/installation)

I connect the board to a USB-C cable connected to my computer.
I downloaded the MicroPython installer and execute it from my Downloads directory (Windows 11)
It detects the board and shows it in the installer app. I click [Install MicroPython]
It briefly says "Flashing firmware...." then errors out.
[image]

fpsst, 2023-07-27:
I have the exactly same problem. Tried to double tap the reset button which makes the green led blink slowly (indicating the bootloader mode, I suppose), but to no avail.

adengle, 2023-07-27:
I found a fix here. Install the Arduino IDE, then add support for the Nano in the boards manager. It appears that this installs the driver to let the board be recognized in Device Manager and then the MicroPython installer will work.

closed 2023-07-27 by adengle