arduino/lab-micropython-installer #20: Add windows driver collection

pillo79, 2023-11-22:
Currently the installer fails on Windows if the PC doesn't have the right drivers (particularly true for ESP32). They are by default installed by the Arduino IDE when adding the "Arduino ESP32 boards" core, but that requirement is unnecessarily complex if one is interested in MicroPython only. 
We publish a ZIP with a set of Windows drivers (for _all_ our boards) that can be downloaded and installed. Would it be possible to add this step to the Windows installer?

The current release is available [here](https://downloads.arduino.cc/tools/drivers-arduino-windows-1.8.0.zip), but it needs to be updated with ESP32 drivers so the version number will be updated shortly.

sebromero, 2023-12-06:
@pillo79 Definitely an option. Do the devices even show up if no driver is installed? I guess they should. I assume there is no easy way to check if the drivers are installed. So there are two possible approaches:

1. Install the drivers in any case before installing MicroPython no matter if they are installed or not
2. Wait for the installation to fail and then install the drivers as an attempt to recover the situation.

Thoughts?

pillo79, 2023-12-06:
> Do the devices even show up if no driver is installed?

I'm not familiar with this Windows API detail, however, the devices should be recognized from an USB standpoint, but missing a driver they would be enumerated as "Unknown device" or something. Are you able to enumerate USB devices similar to what `lsusb` does in Linux? 

I personally prefer option 2 ("Device not found, maybe you are missing drivers. Would you like to install them now?") - even better if you can be sure a known device _is_ there via USB enumeration.