arduino/lab-micropython-installer #22: Add installation instructions (or auto-install) for udev rules on Linux

laurensvalk, 2023-12-19:
It would be nice to display instructions to install udev rules, or perhaps even install them as part of the debian package.

Adding the following rule worked for the Arduino Nano ESP32:
```
# Arduino Nano ESP32 in firmware update mode (DFU mode)
SUBSYSTEM=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="0070", TAG+="uaccess"
```

------------

Out of the box on Ubuntu 22.04 I got the following error, which may be hard to understand for someone who hasn't changed udev rules before.

[image]

sebromero, 2023-12-20:
Thank you @laurensvalk for reporting! We will look into it.

pkvil, 2024-06-01:
I had the same issue on Debian 12, and agree that having instructions would have been very helpful. I ran this script: https://github.com/arduino/ArduinoCore-mbed/blob/main/post_install.sh which worked for me.