arduino/lab-micropython-installer #24: Cannot flash 2nd GIGA connected to the desktop

baqwas, 2024-01-29:
Steps to reproduce:
1. Flash 1st GIGA. No issue
2. Flash 2nd GIGA (_without disconnecting the 1st_). Flash does not complete.
3. Switch off 1st GIGA. 2nd GIGA flashes successfully
4. Repeated the above sequence twice and experienced the same behavior

Haven't examined the DFU-UTIL parameters in detail but Tormond will probably respond quickly at his end when I let him know.

Regards.
The desktop is Ubuntu 23.10 where command line dfu-util worked successfully several months ago when there was no Arduino micropython-installer.

sebromero, 2024-04-25:
Hi @baqwas ! Thank you for pointing this out. This is a use case that makes perfect sense but I didn't consider. dfu-util only checks the product and vendor ID and obviously both GIGA boards have the same so it's undefined which one is used for flashing. I'll come up with a fix and let you know.

sebromero, 2024-07-25:
Should be solved in the latest release.

closed 2024-07-25 by sebromero

baqwas, 2024-07-25:
Thank you very much. Appreciate it since I'm actively needed that functionality right now for some CAN Bus tests!

Regards.