arduino/lab-micropython-installer #25: Not able to flash micropython on Arduino nano 33 ble sense Rev2

saurabh-2905, 2024-02-16:
Hello, I am trying to install Micropython on my Arduino nano, as specified in the title. At first the installer worked great and I was able to run some scripts and play around. Since I am new to this, I was exploring it and tried using different IDEs to see how it works (VSCode, Thonny, Arduino-Lab for Micropython). Everything was working fine. Suddenly, the device froze and I could not upload any files. Later, I was not able to connect to the device. The device was detected, but as soon as I tried to connect to it, it just took ages and did not happen. 

I thought maybe the software is broke and tried to reflash the micropython. I did it directly using the lab-micropython-installer, but it failed. So I thought of doing the whole process again, updating the bootloader, running the Arduino scripts, re-installing the soft-device code, and then flashing Micropython again. But since then it fails every time. In one of the attempt to flash Micropython using this installer, it gave following error:

`Error running bossac: Command failed: "/Applications/MicroPython Installer.app/Contents/Resources/app/node_modules/firmware-flash/bin/darwin/bossac" -d --port=/dev/tty.usbmodem1 -U -i -e -w "/Applications/MicroPython Installer.app/Contents/Resources/app/node_modules/firmware-flash/bin/firmware/SoftDeviceUpdater.bin" -R`

As I said, I am new to this and looking around how to solve this error. Any help would be great. Thanks

sebromero, 2024-04-25:
Hi @saurabh-2905 !
Hmm, interesting. `Uncaught Exception: Error: Cannot create BrowserWindow before app is ready at createWindow` this should never happen. I've added another check but I'm not currently able to re-create.

As for the main issue, could you please add the complete log output to this issue? You can obtain it by starting the app from the terminal (see [here](https://github.com/arduino/lab-micropython-installer?tab=readme-ov-file#-reporting-issues);

Open the Terminal app and type: `"/Applications/MicroPython Installer.app/Contents/MacOS/micropython-installer"` then press enter.

sebromero, 2024-07-26:
If you have any additional info please add it here so I can take a look. Closing this until there is additional data.

closed 2024-07-26 by sebromero