arduino/lab-micropython-installer #34: Supported boards do not match Ardiono site

Jimmeee, 2024-08-30:
Sorry if this is not supposed to be the case (and therefore a dumb question) but the boards listed as being supported here do not match those on the Arduino website. For example, see:

https://docs.arduino.cc/micropython/

The Arduino website says that the Mega is supported. I am trying to get MicroPython on a Rev3 Pro Mini (which is an ATMEGA2560 with the same pinout as the normal Rev3 Mega), and it is not working, either using the instructions there or the .deb file here.

The Arduino IDE finds the board, but the MicroPython Installer does not. 

I know that when I run "micropython-installer" from the command line, it says that Linux is not supported. I guess that means... Linus is not supported. But, since the Arduino pages say it is, I'm confused and so trying all available methods.

sebromero, 2024-11-13:
Hi @Jimmeee ! Thank you for reaching out. None of the ATMEGA boards are supported. It would make no sense since they are not powerful enough to properly run it. The supported Arduino boards can also be seen on the official MicroPython website: https://micropython.org/download/?vendor=Arduino

When looking at our docs website I don't see the Mega it in the list that you mentioned:
[image]
Can you help me find it?

sebromero, 2024-11-20:
Closing this for the time being. Feel free to re-open should the issue persist.

closed 2024-11-20 by sebromero