arduino/lab-micropython-installer #36: MicroPython Installer is not working for Nano ESP32 board

deepwavebill, 2024-12-18:
I get the following error when I try to install MicroPython on my Nano ESP32 board.

FetchError: request to https://downloads.arduino.cc/micropython/index.json failed, reason: unable to get local issuer certificate

I have downloaded the Arduino ESP32 boards package to the Arduino Nano ESP32 and I have run the Blink sketch successfully.

FYI- For Windows the MicroPython installer is named the following: MicroPython.Installer-1.4.0.Windows-Setup.exe.

Thanks for your help.

Bill

ndni, 2025-01-03:
Congratulations @sebromero for the MicroPython project for Arduino!
Given support for NanoESP32 (https://docs.arduino.cc/hardware/nano-esp32/ ESP32-S3), is it compatible with all ESP32 microcontrollers from Espressif (e.g. ESP32-D0WDQ6 (revision 1) see https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf and https://www.keyestudio.com/products/keyestudio-esp32-plus-development-board-woroom-32-module-wifibluetooth-compatible-with-arduino)?
Thanks in advance

sebromero, 2025-01-06:
Hi @deepwavebill I think there were some Certificate issues. It looks like they have been resolved. Can you try it again?

sebromero, 2025-01-06:
Hi @ndni MicroPython itself is compatible with most ESP32 MCUs. You can see the list here: https://micropython.org/download/
Arduino's MicroPython installer tool however for now only supports Arduino boards officially. This is mainly because we rely on the USB Product ID to choose the correct MicroPython image. If you would like to further discuss this, please create a new topic under https://github.com/arduino/lab-micropython-installer/discussions

sebromero, 2025-03-31:
Closing this due to inactivity

closed 2025-03-31 by sebromero