arduino/lab-micropython-installer #38: Micropython-installer error

HPRonhovde, 2025-02-17:
I get the following error when trying to run the micropython firmware installer.

on@3252:~$ micropython-installer
Electron's autoUpdater does not support the 'linux' platform
MESA-INTEL: warning: ../src/intel/vulkan_hasvk/anv_device.c:672: The kernel reported a GTT size larger than 2 GiB but not support for 48-bit addresses
libva error: /usr/lib/x86_64-linux-gnu/dri/iHD_drv_video.so init failed
🔌 No RP2040 device found.
dfu-util: Cannot open DFU device 2341:0070 found on devnum 15 (LIBUSB_ERROR_ACCESS)

Error running dfu-util (stderr): 'dfu-util: Cannot open DFU device 2341:0070 found on devnum 15 (LIBUSB_ERROR_ACCESS)
'
ron@3252:~$ 
I am running a Dell Inspiron 3252 with 8gb ram.  The OS is Linux Mint 22.1 Cinnamon.

I have been using the Arduino IDE to run C++ sketches on an ESP32  Wrover board on a different port but the same USB port.  The installer identifies the EPS32 Nano board in the IDE and I can read the board info via the IDE also.  

Any helpful hints will be appreciated.  THX

sebromero, 2025-03-31:
Hi @HPRonhovde Did you add the udev rules for your board? Please also take a look at this issue: https://github.com/arduino/lab-micropython-installer/issues/22

HPRonhovde, 2025-03-31:
Thanks for the suggestion.  I did not know about any rules for the
installer.  I will try the rules set up and try to load uPython I again.  I
have one more board that needs the firmware updated.