arduino/lab-micropython-installer #39: Memory corrupted after micropython firmware programming Arduino OPTA

panther50500, 2025-03-28:
I'm using Arduino Opta with MicroPython and two of them work fine. However, I ordered one online and encountered a problem after programming the firmware. Files in file explorers (Linux or Windows) are corrupted (special characters displayed and undefined files created). If I try to program it with Arduino, it seems to work fine.
I returned it to the reseller who sent me another one but it has the same problem.
Someone else have this problem ?
Thanks
Frédéric

try with :

ARDUINO_OPTA-20241129-v1.24.1.dfu (which works with another boards)
ARDUINO_OPTA-20250326-v1.25.0-preview.410.gdd7a950bb.dfu (which works with at least one other board)

panther50500, 2025-03-28:
[README.txt](https://github.com/user-attachments/files/19511164/README.txt)

[pybcdc.inf.txt](https://github.com/user-attachments/files/19511170/pybcdc.inf.txt)

[main.py.txt](https://github.com/user-attachments/files/19511173/main.py.txt)

[boot.py.txt](https://github.com/user-attachments/files/19511176/boot.py.txt)

i create extension .txt at boot.py, main.py and pybcdc.inf to add them

sebromero, 2025-03-31:
Hi @panther50500 ! Thank you for sharing this issue. Unfortunately the functionality to transfer files to and from the board (also when reading/writing files) using the file explorer is not very stable. It's a MicroPython issue. We don't recommend doing that. Instead I recommend to either use [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html) or to use our own [Editor](https://labs.arduino.cc/en/labs/micropython) which has file handling capabilities.

panther50500, 2025-03-31:
Hi @sebromero, thank you for your return. I should have given you more details. In fact, I also tested this by unmounting the disk from File Explorer, using the Arduino Lab interface for MicroPython and Thonny.
In both cases, I was unable to copy or read the files; the copy was either impossible, or they were either not present or incorrect.
[image]

facchinm, 2025-03-31:
Cross referencing https://github.com/micropython/micropython/issues/17020 for mainline support (not really IDE related, so I'd close it here)

closed 2025-03-31 by sebromero